COMPASS-DOE/sensor-data-pipeline #300: Add new GCW TR sapflow sensors to design table

stephpenn1, 2025-04-14:
From @aestearns01:
><head></head><div class="elementToProof" style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; font-family: Calibri, Helvetica, sans-serif; font-size: 11pt; color: rgb(0, 0, 0);"><br class="Apple-interchange-newline">Hi all,</div><div class="elementToProof" style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; font-family: Calibri, Helvetica, sans-serif; font-size: 11pt; color: rgb(0, 0, 0);"><br></div><div class="elementToProof" style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; font-family: Calibri, Helvetica, sans-serif; font-size: 11pt; color: rgb(0, 0, 0);">I successfully installed two additional sap flow sensors in the GCW TR zone. TR7 and TR8. The<span class="Apple-converted-space"> </span><a href="https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.google.com%2Fspreadsheets%2Fd%2F11rDawyDAzwQMKQS0dIOsQAHdgHFv4nrV%2Fedit%3Fusp%3Dsharing%26ouid%3D111751932849645868002%26rtpof%3Dtrue%26sd%3Dtrue&amp;data=05%7C02%7Cstephanie.pennington%40pnnl.gov%7Cb0f034758fe446bba94408dd7b69dfc3%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C638802416837361312%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&amp;sdata=K9B2RNR%2BEf43jwDlitjieNM%2FM%2Fl91FrKyGdOhq13jTc%3D&amp;reserved=0" originalsrc="https://docs.google.com/spreadsheets/d/11rDawyDAzwQMKQS0dIOsQAHdgHFv4nrV/edit?usp=sharing&amp;ouid=111751932849645868002&amp;rtpof=true&amp;sd=true" id="LPlnkOWA9cfe5c50-6284-5343-a505-c9af5466177e" class="OWAAutoLink" title="https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.google.com%2Fspreadsheets%2Fd%2F11rDawyDAzwQMKQS0dIOsQAHdgHFv4nrV%2Fedit%3Fusp%3Dsharing%26ouid%3D111751932849645868002%26rtpof%3Dtrue%26sd%3Dtrue&amp;data=05%7C02%7Cstephanie.pennington%40pnnl.gov%7Cb0f034758fe446bba94408dd7b69dfc3%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C638802416837361312%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&amp;sdata=K9B2RNR%2BEf43jwDlitjieNM%2FM%2Fl91FrKyGdOhq13jTc%3D&amp;reserved=0" style="color: rgb(0, 120, 215);">Synoptic SF Monitoring doc</a> and the<span class="Apple-converted-space"> </span><a href="https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.google.com%2Fspreadsheets%2Fd%2F1Y5Q8P_aXhmEX-mCpSsVlozUvxNiraaGhyrTmiC37O6g%2Fedit%3Fusp%3Dsharing&amp;data=05%7C02%7Cstephanie.pennington%40pnnl.gov%7Cb0f034758fe446bba94408dd7b69dfc3%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C638802416837389239%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&amp;sdata=MsPpF8pvXxi0wpg%2BHuzggIRe0Gjp%2FhCrZDr8K8IQ3Yg%3D&amp;reserved=0" originalsrc="https://docs.google.com/spreadsheets/d/1Y5Q8P_aXhmEX-mCpSsVlozUvxNiraaGhyrTmiC37O6g/edit?usp=sharing" id="OWA9d17a548-f7d5-2e60-35e3-72d426ed4851" class="OWAAutoLink" title="https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.google.com%2Fspreadsheets%2Fd%2F1Y5Q8P_aXhmEX-mCpSsVlozUvxNiraaGhyrTmiC37O6g%2Fedit%3Fusp%3Dsharing&amp;data=05%7C02%7Cstephanie.pennington%40pnnl.gov%7Cb0f034758fe446bba94408dd7b69dfc3%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C638802416837389239%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&amp;sdata=MsPpF8pvXxi0wpg%2BHuzggIRe0Gjp%2FhCrZDr8K8IQ3Yg%3D&amp;reserved=0" style="color: rgb(0, 120, 215);">CB Tree Inventory</a> have both been updated to reflect these additions. Please let me know if you need any additional information for the L1 data pipeline.

Site | Plot | Sapflux_ID | Tree_ID | Logger | Port | PakBus | Installation_Date | Last_Checked | Status | Pending_Action | Notes
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
GCW | TR | 7 | 916 | 432 | 7 | 432 | 12-Apr-2025 | 12 Apr 2025 | Active |   |  
GCW | TR | 8 | 917 | 432 | 8 | 432 | 12-Apr-2025 | 12 Apr 2025 | Active |   |  

This is related to #284

bpbond, 2025-04-15:
Overlaps with #284

bpbond, 2025-06-24:
Done by @stephpenn1 in #332

closed 2025-06-24 by bpbond