COMPASS-DOE/sensor-data-pipeline #326: Feedback on L2 file structure

bpbond, 2025-06-20:
Hello @stephpenn1 if you have time I would appreciate feedback on our current L2 file structure as currently in the #324 PR:

[image]

Columns "TIMESTAMP", "Site", "Plot", "Instrument", "Instrument_ID", "Sensor_ID", "Location", and "research_name" are all as in L1.

L2-specific columns:

| Column | Description |
|--------|--------|
| Value | Usually this will be the same as L1, but because in L2 data timestamps are _always_ 15 minutes apart, during e.g. TEMPEST floods this will be the average of multiple values. |
| n | This is the number of L1 values averaged to produce this L2 value. If zero and a value is present, that means that linear interpolation was used to fill a short gap. |
| n_drop | Number of values dropped _before_ averaging for being out of bounds, out of service, of failing QA/QC (outlier testing). |
| GF_MAC | Gap-filled time series, based on the _mean annual cycle_. In the screenshot above, you can see in the red circle values that were gap filled in this way, because of missing data. In this case, the average of sensors 1-3 was used, since there's a gap in sensor 4.  |

----> Does this make sense to you? Are there changes in definition, nomenclature, anything that would be useful? Thank you

stephpenn1, 2025-06-20:
Do we have a list of variables we plan to do GF-MAC for?

stephpenn1, 2025-06-20:
>every variable in L2
Okay, this may cause weird jumps in something like the TEMPEST flood data no?

bpbond, 2025-06-20:
Hmm I don't see why 😕 -- but maybe I'm missing something.

stephpenn1, 2025-06-20:
> Hmm I don't see why 😕 -- but maybe I'm missing something.

Let's chat about it at our meeting next week, maybe im just not understanding!

bpbond, 2025-06-20:
>n_avg?

That seems a little confusing to me. What about `n_vals` or something like that, i.e. number of values that were averaged to produce this L2 value?

bpbond, 2025-06-23:
>Okay, this may cause weird jumps in something like the TEMPEST flood data no?

Oh, now I see your point 💡 

Yes, absolutely, true, and it'll affect gap-filling in non-flood years. Hmm

stephpenn1, 2025-06-23:
Yes exactly, we'll have to think

closed 2025-06-26 by bpbond