COMPASS-DOE/sensor-data-pipeline #333: CRC UP replaced trees for DLG

stephpenn1, 2025-06-24:
Removed CRC UP sensors from `Compass_CRC_UP_303` logger:

Site | Plot | Sapflux_ID | Tree_ID | Logger | Port | PakBus | Installation_Date | Last_Checked | Deactivation_Date
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
CRC | UP | 01D | 301 | 303 | 9 | 303 | 16-Mar-2022 | 10-Jun-2025 | 10-Jun-2025
CRC | UP | 02D | 302 | 303 | 10 | 303 | 16-Mar-2022 | 10-Jun-2025 | 10-Jun-2025
CRC | UP | 06D | 306 | 303 | 11 | 303 | 7-Jul-2022 | 10-Jun-2025 | 10-Jun-2025
CRC | UP | 08D | 308 | 303 | 12 | 303 | 16-Mar-2022 | 10-Jun-2025 | 10-Jun-2025

Installed CRC UP sensors in their place in `Compass_CRC_UP_303` logger:

Site | Plot | Sapflux_ID | Tree_ID | Species | Logger | Port | PakBus | Installation_Date
-- | -- | -- | -- | -- | -- | -- | -- | --
F4 | 9 | 356 | QURU | CRC-UP | CRC_UP | 9 | 303 | 10-Jun-2025
F4 | 10 | 351 | QURU | CRC-UP | CRC_UP | 10 | 303 | 10-Jun-2025
F4 | 11 | 343 | QURU | CRC-UP | CRC_UP | 11 | 303 | 10-Jun-2025
F4 | 12 | 339 | QURU | CRC-UP | CRC_UP | 12 | 303 | 10-Jun-2025

@aestearns01

stephpenn1, 2025-06-24:
@bpbond should we remove these for CRC for v2 and wait to add the DLG sensors until next release?

bpbond, 2025-06-24:
Hi @stephpenn1 I don't think there's any problem with making these changes before v2, we still have time, but your call.

stephpenn1, 2025-06-25:
Re-opening this to hand DLG sensors for the next release

bpbond, 2025-06-25:
Whoops sorry!