COMPASS-DOE/sensor-data-pipeline #335: Compare L2 gw-wl-below-surface with expert expectations

bpbond, 2025-06-24:
In #330 @peterregier said

>t would be good for us to take values calculated by the pipeline for all sites for the last week and compare to my synoptic check values just to triple-check that things are matched up.

Agreed. As a quick sanity check, here are L2 `gw-wl-below-surface` plots for MSM:

[image]
[image]
[image]

bpbond, 2025-06-24:
...and TMP saltwater:

[image]

peterregier, 2025-06-25:
[image]
[image]

MSM and CRC values seem to match!

bpbond, 2025-06-25:
Woot! Another entry in the "okay @bpbond didn't eff it up" column. Thank you!