COMPASS-DOE/sensor-data-pipeline #346: Are redox probe bounds correct?

bpbond, 2025-06-28:
The instrument bounds for redox probes are listed as (-1000, 1000) mV, but many of the probes spend a lot of time outside of this range:

[image]
[image]

@roylrich Are we sure about these bounds? All the out-of-bounds data, in red, are dropped in Level 2.

roylrich, 2025-06-29:
I think I need to check MV range. The differential channel may range differently.

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology

External Email - Exercise Caution

TMP_C_20240507-20241231_redox-35cm_L1_v1-0.png (view on web)<https://github.com/user-attachments/assets/9502338b-85cd-453f-8e37-ae3b48455ee9>
TMP_F_20240507-20241230_redox-10cm_L1_v1-0.png (view on web)<https://github.com/user-attachments/assets/c1c2319b-c469-4fe5-a8dd-c42668cbd9ba>

roylrich, 2025-06-29:
The differential channel will record data between -2500 and 2500 mv

SubScan(2000,msec,20)
PulsePort(2,5000)
'Switch to next AM16/32 Multiplexer channel
''VoltDiff (RedoxRa(LCount),1,mv2500C,1,True,0,_60Hz,1.0,0)
VoltDiff (RedoxRa(LCount),1,mv2500C,1,True,0,_60Hz,1.0,0)
LCount=LCount+1

bpbond, 2025-06-29:
Beautiful! Thank you! I'll fix the bounds.

@stephpenn1 for awareness

PR #350 (2025-06-30): Redox bounds adjustment
Closes #346 

Thanks @roylrich !
closed 2025-06-30 by bpbond