COMPASS-DOE/sensor-data-pipeline #347: Problem with CRC and OWC sapflow wiring or programs?

bpbond, 2025-06-28:
The CRC and OWC sapflow values are way out of bounds, and as a result all the data are being dropped in L2. No other site looks like this.

[image]
[image]

roylrich, 2025-06-29:
I am not sure what I am seeing here... Is this just two trees or representative, do we have a unit issue here?
RR

bpbond, 2025-06-29:
You're seeing values out of range by 10^3. Here's TEMPEST, for example

[image]

roylrich, 2025-06-29:
I just looked at sapflow on both these loggers, the values are mostly NAN at this point but the ones that are showing are in range.  am not going to look at the loggernet tables. _RR

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology

External Email - Exercise Caution

CRC_TR_20240101-20241231_sapflow-3.5cm_L1_v1-0.png (view on web)<https://github.com/user-attachments/assets/c9e4c92d-c7dd-4af6-9dd1-dd15f69ca9c9>
OWC_UP_20240101-20241231_sapflow-3.5cm_L1_v1-0.png (view on web)<https://github.com/user-attachments/assets/0eb70285-d1d2-4fd2-8c5e-07e4cbb8ebcc>

roylrich, 2025-06-29:
[cid:2e5d0e3f-69d5-48c9-8496-d8a64ab3754d]

CRC Transition has only one live sensor currently.

and its been this way since for a long time... someone not checkings system at all.
OWC is similar, very few senors, it looks like they start to very high in range as they start to lose function. There are a few sesnors at OWC that are still in range.
RR

OWC Upland

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology