COMPASS-DOE/sensor-data-pipeline #359: PTR has unrealistic salinity in L2 data

bsulman, 2025-07-11:
PTR site has salinities > 100 in L2 groundwater data. I think this is inherited from the L1 data. So I'm not sure if this is a known issue at this point (i.e. no screening of values is included in L2, just gap filling) but I wanted to point it out. See the PTR panel on the lower row in this figure:

[image]

bpbond, 2025-07-11:
Oh, wow. Thank you @bsulman ! 

Looks like this is driven entirely by a period in 2023 when the data went crazy:

[image]

The current bounds for `gw-salinity` are (0, 350) which explains why it's not getting filtered out.

Other than than period, things look mostly OK:

[image]

@Fausto2504 any insight on what happened in 2023?

bpbond, 2025-07-13:
Bad data start 2023-08-01 and ends 2023-12-05

Weirdly, other data look OK during this period:

[image]

[image]

nickdward, 2025-07-14:
Note that this issue results in a non-calculatable water level in the L2 dataset. Here's a quick slide deck on how the wrong salinity data influences gap filling and also L2 calculation

[Portage GW salinity issue.pptx](https://github.com/user-attachments/files/21221565/Portage.GW.salinity.issue.pptx)

Fausto2504, 2025-07-14:
I am in contact with InSitu, attempting to access the raw conductivity data and understand conversions.

bpbond, 2025-10-23:
@stephpenn1 points out we could also solve this by marking these data F_OOS (out of service)

bpbond, 2025-11-03:
For v2-1, we need to decide the action here. I believe that these are the options:

1. Mark the instrument as out of service 2023-08-01 to 2023-12-05. The data will still appear in L1 but be flagged (`F_OOS`), and will be removed from L2
2. Back-correct the raw data by removing (`NA`) the problematic salinity data -- conductivity, salinity, density, and TDS, per @Fausto2504 's comment above
3. Back-correct the raw data with a correction factor to 'save' the data

bpbond, 2025-11-09:
@stephpenn1 

>this also comes down to what our role should be in the QA/QC process. In my opinion, we should avoid editing and by-hand corrections on raw data values wherever possible

Insightful point, and I agree in principle. But at the same time, here's a case where marking out-of-service means we lose four months of data from the entire Aquatroll (19 separate variables) because of a calibration error involving just _one_ of those variables. I.e., all the other data are good. So that makes me hesitate.

@nickdward 

>Option 1 makes sense for L1, and for L2 either remove or include the back correction

L2 is built from the L1 data, so we can't have different solutions for each level.